bigtreetech/Pad5 #1: GPIO pinout

cakeslob, 2022-03-26:
Can you provide some kind of pinout document or some label for the 40 pin GPIO? A document would be helpful, thank you.

jrgGearbox3d, 2022-04-05:
I am also looking for this.  The manual states to hook 5V to Pin 40, which is not typically the GPIO layout for a CM4 / PI.  Based on this I would guess that the pins are numbered backward from a typical 40PIN PI GPIO.

It would be nice to see the real GPIO and circuit before smoking a nice IPS screen.

bigtreetech, 2022-04-07:
Hello, https://github.com/bigtreetech/Raspberry-Pad/commit/2a0127b8a28db4e08ce698afd2b463529ae75a97 have updated the the pin out description of 40 pin GPIO on manual, Please check.

closed 2022-04-10 by cakeslob