bigtreetech/Pad5 #3: Control Display Backlight?

lalten, 2022-04-15:
Hi, how can I control the Raspberry Pad 5's display backlight and power?

I'm on `2022-04-04-raspios-bullseye-arm64-lite` with Kernel 5.15.32-v8+, running KlipperScreen (installed via [kiauh](https://github.com/th33xitus/kiauh))

```
❯ cat /boot/config.txt | grep "^\w"
dtparam=spi=on
dtparam=audio=on
camera_auto_detect=1
display_auto_detect=1
dtoverlay=vc4-kms-v3d
max_framebuffers=2
arm_64bit=1
disable_overscan=1
otg_mode=1
arm_boost=1
dtoverlay=dwc2,dr_mode=host
dtparam=i2c_vc=on
dtoverlay=i2c-rtc,pcf8563,i2c_csi_dsi
```
`/boot/dt-blob.bin` is also installed as per the manual.

A sysfs entry is present:
```
❯ ls /sys/class/backlight/10-0045/
actual_brightness  bl_power  brightness  device  max_brightness  power  scale  subsystem  type  uevent
```
But sending commands has no effect. (using [rpi-backlight](https://github.com/linusg/rpi-backlight)):
`sudo rpi-backlight -p off /sys/class/backlight/10-0045`
Display stays on and at 100% brightness...

Darkgoppi, 2022-09-01:
I have the same behavior as described by lalten. I tried to switch the back light power by manipulating the `bl_power` file in `/sys/class/backlight/10-0045/` based on the status of xscreensaver. Is there any way to control the back light brightness and power?

Darkgoppi, 2022-09-10:
Hi, I contacted the BIQU support on this issue/question. This is the answer from them:

"Sorry, no. This is a constant light that the screen lights up."

So it seems, that it is not intended to control the backlight at all, which is a pity.

sl1pkn07, 2023-04-23:
and via buttons hardware? (three in the corner)

kbenyous, 2024-03-28:
> Hi, I contacted the BIQU support on this issue/question. This is the answer from them:
> 
> "Sorry, no. This is a constant light that the screen lights up."
> 
> So it seems, that it is not intended to control the backlight at all, which is a pity.

Hello,
According to their technical docs ( schematics https://github.com/bigtreetech/Raspberry-Pad/blob/master/Pad5/Hardware/BIGTREETECH%20Raspberry%20pad%205-SCH.pdf ) the 3 hardware buttons (BRI+, BRI- and LCD_DIR) are plugged to some pins of a ARM Cortex M0 MCU. So it could be software-controlled, if BTT would implement it in the firmware of this MPU !