bigtreetech/Pad5 #4: does not show on network

Machine73, 2022-05-15:
Purchased 2 pad5's and neither will connect to network. SD cards with os installed work perfectly in a standard PI. But once installed in the PAD5 no connection,

othermod, 2022-05-15:
It is the wired LAN port you're having trouble with? I've been using one of these for a while and I have it wired to a decent network switch that works fine with it. I recently tried using it with a cheap switch, and it can't get an IP address with it connected to that switch. Running ifconfig just shows a standard 169.254 address.  Just curious whether we are having the same problem.

Machine73, 2022-05-15:
No. Over wifi. And they are the CM4 w/ wifi.

blooest, 2022-05-18:
You haven't really given us any information. My best guess is that you got a CM4 with eMMC that will not boot at all from SD. Connect an HDMI display so you can see what's going on with it. Other than that, the CM4 wifi has nothing at all to do with the Raspberry Pad so you're probably better off seeking information from a more generalized/larger RPi community.

@othermod To note, I'm here chasing the same problem. The switch I'm working with is a known-good Unifi box, though. It's detecting the client, providing an activity light on its end, but I get a whole lot of nothing on the Pi side and then a 169.254 zeroconf address. Was beginning to think the port was dead, but I guess I'll have to move things around and try another switch. Maybe we should open a separate issue?

jstebbins, 2022-08-19:
Confirming same behavior as @blooest. I am also connected to a Unifi switch (model US-8-60W). Fails to link at GB speed, but forcing link to 100BASE-T works.

When failing to link at GB speed, I see an initial attempt to bring the link up in the system logs, then the link goes down and does not connect.  The green light on the ethernet connector of the Raspberry Pad blinks at a slow regular interval.

blooest, 2022-08-19:
@jstebbins Sorry, I actually never circled back to this thread -- BTT support actually did solve that issue for me; in the end, I had to [physically desolder three SMD components on the board](https://imgur.com/a/lClHPaa), but the gigabit PHY worked flawlessly afterwards. Hope that helps!

jstebbins, 2022-08-19:
@blooest Thanks! Did they provide any additional details regarding the nature of the problem? Do those components affect any other functionality? Would hate to fix one thing only to render something else non-functional :grin:

blooest, 2022-08-19:
@jstebbins They weren't _too_ particular to me as an end user, but as long as you _only_ touch those three, it shouldn't affect anything else. They're resistors across the traces from the Magjack to the breakout for the PHY on the CM4. They don't physically interact with any other components. From what I've inferred (read: **I'm just another customer making educated guesses**), they have/had multiple suppliers for the Magjack itself, and the ones they prototyped with didn't have resistance built in, which led them to add these resistors, but the ones they shipped, at least to the two of us, do, and that extra resistance is causing this issue.

jstebbins, 2022-08-19:
> the ones they prototyped with didn't have resistance built in, which led them to add these resistors, but the ones they shipped, at least to the two of us, do

Ahh, that makes sense. Seen that happen before. Thanks for passing along the info.

jaymcd, 2022-10-25:
I have no wifi either - error on boot says bcrm_pci link down

FriendlyNGeeks, 2024-07-26:
> @jstebbins Sorry, I actually never circled back to this thread -- BTT support actually did solve that issue for me; in the end, I had to [physically desolder three SMD components on the board](https://imgur.com/a/lClHPaa), but the gigabit PHY worked flawlessly afterwards. Hope that helps!

I cant believe this worked, but it makes sense from a prototype!!!! Nerve wrecking getting those itty bitty resistors