bigtreetech/Pad5 #7: CM4 2gb lite version, raspios, and usb ports

jaymcd, 2022-08-02:
I have the PAD 5 - I finally managed to get a CM4 ( 2GB lite version) I managed to get RaspiOS ( Raspian) installed. Even after following the instructions in the manual - including the overlay stuff I cannot for the life of me get the USB ports to be usable. Wondering what I am missing

OAltr, 2022-08-17:
Hi @jaymcd I had the exact same problem. What fixed it for me was setting the BOOT and BOTG switches to their off position. Had them on because I previously used a CM4 with eMMC.

brunoAmado, 2022-09-03:
installng [Fluiddpi 1.19](https://github.com/fluidd-core/FluiddPI/releases/tag/v1.19.0) using LITE Version(Micro SD card)
ssh already in image
add wpa credencials in fluiddpi-wpa-supplicant.txt
configure USB 2.0 Hub in config.txt file
verification of microswith (all off)
result:
Red light and Blue but no Green :(

Did I miss something ?

pl4nkt0n, 2022-09-04:
As described in the data sheet of the CM4 from Pi-Foundation and in the manual from BigTreeTech:

"With the Raspberry Pi OS the following has do be done to activate the USB-Ports for the CM4: To enable the USB 2.0 ports on the Compute Module 4, you need to edit the boot config file (/boot/config.txt) and add with the last row:

dtoverlay=dwc2,dr_mode=host

It is deactivated by default in the OS of RaspiOS to save energy.

brunoAmado, 2022-09-05:
Hi,

As I said in the last mail. I did the configuration as explained in the
BigTreeTech Manual.
Please see the attachments and video.

https://drive.google.com/file/d/1MA2Fw5SE6T7sgGQLoNIgWznggDXRX3d6/view?usp=sharing

Best Regards,

Bruno

---------- Forwarded message ---------

usb ports (Issue #7)
To: bigtreetech/Raspberry-Pad ***@***.***>
Cc: Bruno Pires Amado ***@***.***>, Comment <

# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
# enable raspicam
start_x=1
gpu_mem=128

brunoAmado, 2022-09-06:
```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
# enable raspicam
start_x=1
gpu_mem=128
dtoverlay=dwc2,dr_mode=host

```

jaymcd, 2022-10-20:
still cannot get by usb ports to work. I even sent my cm4 back and they sent a new one... ???
pios installed screen working but getting usb to work - not so much. I am usb the USB-C for power does that matter ? It is plugged into a powered USB.

OAltr, 2022-10-20:
Just noticed this section in you config:

> [cm4]
> # Enable host mode on the 2711 built-in XHCI USB controller.
> # This line should be removed if the legacy DWC2 controller is required
> # (e.g. for USB device mode) or if USB support is not required.
> otg_mode=1

I'll have to check on my setup tomorrow, but I don't remember a cm4 section in mine. Maybe removing the org_mode=1 could help.

In the past I had a problem when adding any commands below a section (e.g. [cm] or [all]) and had to add it above any of those (but this might be a non issue and it used to be a different problem back then 😄)

jaymcd, 2022-10-25:
No USB or Wifi. Moved my CM4 over to a CMIO module board - USB works fine but not WIFI ( obviosuly a broadcom driver issue).
There is either something wrong with my PAD 5 or the instructins are incorrect. I have tried everything suggested to no avail. Anyone havea working image?